JaneliaSciComp/G4_Display_Tools #22: Main Host application memory error

taylorlm88, 2021-07-19:
If you try to send a command to the Main Host application via its Labview GUI before it has been connected with the command connectHost, it starts sending a slew of "out of memory" errors. This error window pops up over and over again, you might get a dozen or more before you manage to close the application. It'd be nice to have a better way of handling this if a user forgets to run the 'connectHost' command, like a single error message reminding them.

floesche, 2022-03-01:
To replicate this error, start `G4 Host.exe`, then issue the command `Set Pattern ID` = 1. This leads to a quick increase of memory usage and the application crashes with an out of memory error within a few seconds.

The pattern can either be set via the GUI (see below) or via the TCP command [0x03 0x03 0x01 0x00].

[image]

floesche, 2022-03-01:
Sending a `Set Pattern and Position Function` with any number (eg `[1 1]`) also creates a similar memory issue.

[image]

floesche, 2022-03-01:
This is what `G4 Host.exe` reports:

[image]

floesche, 2022-05-16:
This problem is fixed in v240

closed 2022-05-16 by floesche