JaneliaSciComp/G4_Display_Tools #46: Unexpected frames being logged for short period after 'start-display' command is received

taylorlm88, 2022-02-03:
Part of the Log, Log.Frames.Position, keeps track of which frame of a pattern is being displayed at any given time. It is accompanied by Log.Frames.Time which takes a reading every 2 ms. We use this to figure out when exactly a pattern started moving on the screen by looking for the first time after the 'start-display' command is received that the frame position changes. This is not working, however, because approximately the first 10 (20 ms worth) frame position readings after the 'start-display' command are never 0 as expected, but are other random numbers, as if it is still getting readings from the previous trial. We need to understand what exactly is being logged here - whether the frame numbers being logged are actually being displayed on the screen, and what exactly happens when a pattern is told to change via the 'start-display' command. Is there just an expected delay between the arena registering the 'start-display' command and the screen actually updating the pattern? If so, what causes this delay and is there a way we can measure it in order to make more accurate estimates of pattern movement time?

mbreiser, 2024-04-23:
presumably addressed in #49 - to be verified

taylorlm88, 2024-06-25:
Confirmed this issue seems to have been resolved. I don't see any unexplained frame position numbers at the start of the trial anymore.

closed 2024-06-25 by taylorlm88