JaneliaSciComp/G4_Display_Tools #50: First Analog Input channel

taylorlm88, 2022-02-22:
Currently, when we activate all four analog input channels to stream back data while the screen is running, we get back the following data: 

channel 1 - LmR
channel 2 - Left wing
channel 3 - Right wing
channel 4 - Wing beat frequency

Would it be possible to change channel 1 so instead of LmR, it returns the frame position data being logged in Log.Frames.Position? Receiving LmR data here is a bit redundant, since we have the left and right wing data separately, and we could use the frame position data in real time.

floesche, 2022-03-29:
Feedback received via email:

> For this request, you have two positions (X Y) that are 16 bit, but only one channel to put those values in. This could work if I make the upper 8 bits X and lower 8 bits Y, but only if there are less than 256 positions for each.

floesche, 2022-03-29:
Response (via email):

"it would be sufficient to get the  X position of the frame. "

taylorlm88, 2022-04-25:
Yes! ADC0

floesche, 2022-05-13:
Received via email (as a description for v240):

> Build with x position logging for ADC0 ch.

[G4 Host v240](https://github.com/floesche/LED-Display_G4_Display-Tools/releases/tag/host-v240)

floesche, 2022-08-05:
This seems to work well.

Data:
[X-Position-of-Frame.xlsx](https://github.com/JaneliaSciComp/G4_Display_Tools/files/9269171/X-Position-of-Frame.xlsx)

DAC0 channel:
[image]

closed 2022-08-05 by floesche