JaneliaSciComp/G4_Display_Tools #54: 'stop_log' command sometimes not being received by Main Host

taylorlm88, 2022-02-23:
Summary: The ‘stop_log’ command is not received by the Main Host application.

Description: Sending ‘stop_log’ (0x01, 0x40) does not always stop the log file and seems to be ignored. There is no feedback from the Main Host if the logging was closed successfully. 
Additional / related question: Is this a symptom of an underlying timing-related issue – e.g., are some commands ignored if sent in a certain order? 

code to replicate this issue found in test5.m at https://gist.github.com/floesche/874531bf83aba604a747d9d92e7bca68#file-test5-m and also included below: 

```matlab

%% script to demonstrate a problem

clear all;

connectHost;

Panel_com('change_root_directory', 'C:\matlabroot\G4');
Panel_com('start_log');

Panel_com('start_display', 0.5);
pause(1);
Panel_com('stop_display');
pause(0.1);

allOn = Panel_com('all_on');
pause(0.1);
allOff = Panel_com('all_off');

% This stop log isn't received by G4 Host unless there is an additional
% pause added: 
% pause(0.2)
hasLogStopped = Panel_com('stop_log');
disconnectHost;

```

floesche, 2022-02-23:
I cannot replicate this issue with the following Python code. This suggests it's not a "Main Host" issue, but rather somewhere on the Panel_com or PanelController side.

```python
import socket, time

HOST = 'localhost'
PORT = 62222

def msnd(sck, snd, verbose=True):
    sck.sendall(snd)
    if verbose:
        rspns = sck.recv(1024)
        print(f"Response to '{snd}' is '{rspns}'")
    
def get_cmd_rootdir():
    root_directory_name = 'C:\matlabroot\G4' # actual path
    root_dir = root_directory_name.encode('utf-8')
    dir_length = len(root_dir)
    my_command = b'\x43' + dir_length.to_bytes(2, 'little')
    return my_command + root_dir

with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as s:
    try:
        s.connect((HOST, PORT))
        
    except ConnectionRefusedError:
        print(f"G4 Host doesn't appear to be running on {HOST}:{PORT}")
    print(f"Successfully connected")
    
    # Set dir
    msnd(s, get_cmd_rootdir())
    # Start Log
    msnd(s, b'\x01\x41')

    # Start Display 0.5
    msnd(s, b'\x03\x15' + int(.5*10).to_bytes(2, 'little'))
    time.sleep(1)
    msnd(s, b'\x01\x30')
    time.sleep(0.1)

    # All on
    msnd(s, b'\x01\xff')
    
    time.sleep(0.1)
    
    # All Off
    msnd(s, b'\x01\x00')
    # Stop Log
    msnd(s, b'\x01\x40')

```

taylorlm88, 2023-10-13:
Pretty sure this has been resolved with last years updates removing Panel_com and using the panels controller instead.

closed 2023-10-13 by taylorlm88