JaneliaSciComp/G4_Display_Tools #55: Create command to log SPI responses from Display

taylorlm88, 2022-02-23:
Summary: Log SPI response from Display.

Description: For debugging purposes we would like to record the SPI responses that the Main Host receives from the panels. In most cases we expect the   message that is sent to the panel is simply sent back to confirm it is working. Would there be any way to log these messages? Even if the SPI messages are logged only for a single panel, this may help us debug times when the screen stops displaying but we do not understand why. One idea to implement would be two additional commands ‘Start Debug’ and ‘Stop Debug’ that mirror the behavior of ‘Start Log’ and ‘Stop Log’ but provide more verbose logging. For example, ‘Stop Debug’ could be [0x01 0x50] and ‘Start Debug’ could be [0x02 0x51 0xRC ] with R being a row number, C being a column number for the panel we want the SPI response logged. In either R or C case, "0" would be a catch all (all in that row, all in that column…)

achiusciotex, 2022-05-26:
Currently don't have any DMA available to stream this data back.  Also the SPI Error indicator on the front panel shows constant errors so I don't think the SPI responses are working properly. I think the blank frames issue is not related to the SPI and more related to the data not being available to send to the arena.

floesche, 2022-05-26:
Would it be possible to repurpose one of the DMAs for a specific command, eg the mentioned Start Debug? 

The SPI errors and issues with our Firmware are actually the reason we want to log this SPI response.

achiusciotex, 2022-05-26:
The only 2 DMA channels I could repurpose are either the ADC DMA or the Log DMA(frame times, Treadmill, AO Func). I think repurposing the ADC DMA is the better option. I'll try to make a separate build first to see if it works.

achiusciotex, 2022-05-27:
Would logging a complete column be sufficient for now? Logging specific panels and rows would take a bit more work.

achiusciotex, 2022-06-03:
I'll go ahead and implement the single column feature. I'll expand it to row and single panel if that is not sufficient at a later point.