JaneliaSciComp/G4_Display_Tools #61: Frame Position and Frame time TDMS files are empty

floesche, 2022-05-16:
Every once in a while, the log files related to the frame are empty. On one of the setups with G4 Host v235 this happens as often as once every other day (sample from early to mid April 2022).

One example of the out generated is attached. The error became visible during the `alignFly_LPTC` script, which generated the following output (but this is not the cause, rather the moment when the problem was detected).

[SS02103_JFRC49-16_27_53.zip](https://github.com/JaneliaSciComp/G4_Display_Tools/files/8701362/SS02103_JFRC49-16_27_53.zip)

```matlab
alignFly_LPTC

ans =

     1

ans =

  struct with fields:

         numByte: 0
         success: 0
     commandCode: 0
    responseData: ''

ans =

  struct with fields:

         numByte: 0
         success: 0
     commandCode: 0
    responseData: ''

Do you want to repeat the alignment? Y/N : y

ans =

  struct with fields:

         numByte: 0
         success: 0
     commandCode: 0
    responseData: ''

ans =

  struct with fields:

         numByte: 0
         success: 0
     commandCode: 0
    responseData: ''

Do you want to repeat the alignment? Y/N : y

ans =

  struct with fields:

         numByte: 0
         success: 0
     commandCode: 0
    responseData: ''

ans =

  struct with fields:

         numByte: 0
         success: 0
     commandCode: 0
    responseData: ''

Do you want to repeat the alignment? Y/N : y

ans =

  struct with fields:

         numByte: 0
         success: 0
     commandCode: 0
    responseData: ''

ans =

  struct with fields:

         numByte: 0
         success: 0
     commandCode: 0
    responseData: ''

Do you want to repeat the alignment? Y/N : n
Starting parallel pool (parpool) using the 'local' profile ...
Connected to the parallel pool (number of workers: 8).
Error using G4_TDMS_folder2struct (line 63)
Unable to perform assignment because the indices on the left side are not compatible with the size of the right side.

Error in G4_conductor_controller/run (line 786)
            G4_TDMS_folder2struct(fly_results_folder);

Error in G4_conductor_view/run_exp (line 619)
            self.con.run();

Error in G4_conductor_view>@(varargin)self.run_exp(varargin{:}) (line 139)
                'units', 'pixels', 'Position', [15, self.fig_size(4)- 305, 115, 85],'Callback', @self.run_exp);
 
Error using alignFly_LPTC (line 59)
Error while evaluating UIControl Callback.
```

floesche, 2022-05-16:
Hard to replicate. Is there a quick fix? If not, it's not worth the time trying to fix this issue…

achiusciotex, 2022-05-26:
I think we should table this for now because this will be very hard to fix without it being easily replicated.