JaneliaSciComp/G4_Display_Tools #65: Closed Loop Stripe Position plot in DA_report is empty

floesche, 2022-12-07:
Very rarely, the 3rd plot on the first page ("Closed Loop Stripe Position") looks like this:

[image]

This happens rarely. This example is from fly SS03188-358 recorded on 2022-12-02T16:41:33. Data should be on the flight arena computer.

I think we had seen this issue before, but I couldn't find a reference to it.

taylorlm88, 2022-12-22:
This issue was fixed, or as fixed as it can be, in commit [9bc609d](https://github.com/JaneliaSciComp/G4_Display_Tools/commit/9bc609dcc91a053d201d216f9b8ca71d2a60a00c)

Before this update, we got the expected number of data points of an inter-trial by looking at the first inter-trial's data, because they are all more or less the same. When we then get the average line to plot by averaging all the inter-trial data together, we divide by that expected number of data points. When a graph like this turns up, it was because the first inter-trial got messed up - the screens were dark and not displaying - so the number of data points displayed was zero. Thus we ended up dividing by zero. 

The code has been changed in the above commit to get the number of data points in all inter-trials throughout the experiment and average the nonzero ones together in order to get the expected number of data points. So it stands to reason that the only way to get a plot that looks like this is if none of the inter-trials ran properly. 

I cannot find the data referenced above. I can't find a folder called SS03188-358, and I looked through the data from 12/2 - 12/7 and couldn't find a fly with this error, so I haven't been able to look at the data to confirm. I expect you'd find the inter-trial data is empty. If you can point me to the exact location of the data, I can look at it and see if this is true. If it is, there's not much I can do in the code to fix it except maybe code in a message to the user telling them if this happens, so at least they know why their plot is wonky.

taylorlm88, 2022-12-22:
commit 7eebc3057c1aa3fbc702e9c5f22a2187873a9a2c adds a message that is displayed to the user in this rare scenario alerting them that their Stripe Fixation histogram may not plot properly because the inter-trials may not have displayed.

floesche, 2022-12-22:
Great, I knew you had addressed this before.

I did a file search for `SS03188`  and one of the folders that showed up was inside the LPTCscreen directory the `Bad Data` folder which contains the data set (`SS03188_JFRC49-16_41_33`).

I also pulled your latest patch to the computer…

taylorlm88, 2022-12-22:
Great, thanks, I’ll go back to that directory and check the data. I searched for SS03188-358 so that’s probably why I couldn’t find it.

closed 2023-10-12 by taylorlm88