JaneliaSciComp/G4_Display_Tools #67: Wrong hard-coded path in `create_settings_file()`

floesche, 2022-12-21:
I am trying to follow the documentation on how to run a data analysis manually. The attempt to [create a data analysis settings file](https://reiserlab.github.io/Modular-LED-Display/Generation%204/Display_Tools/docs/data-handling_analysis.html#creating-your-data-analysis-settings-file) fails with `<mypath>` is the directory that contains the `metadata.mat` file produced by the conductor:

```matlab
> create_settings_file('newanalysis.mat', '<mypath>')
Error using load
Unable to read file '/Users/taylorl/Desktop/narrowGrating09-08-21_17-27-53/later_processing_settings.mat'.
No such file or directory.

Error in DA_plot_settings (line 484)
    process_settings = load(exp_settings.path_to_processing_settings);

Error in create_settings_file
```

floesche, 2022-12-21:
OK, I understood now that the step above wasn't necessary for my use case and I can skip it.

I already have a file `single_fly_analysis.mat` in my folder, which should be OK as a basis for the analysis. According to the documentation of [Running a typical analysis](https://reiserlab.github.io/Modular-LED-Display/Generation%204/Display_Tools/docs/data-handling_analysis.html#running-a-typical-analysis) I now saw this error: 
```matlab
>> da = create_data_analysis_tool('<mypath>\single_fly_analysis.mat', '-single', '-hist', '-tsplot');`
Error using create_data_analysis_tool (line 195)
cannot find processed file in specified folder
```

I am guessing the problem is that I moved the file to a different location now? So, unless I move the files back to where they were before, I would probably need to go back to running the `create_settings_file()` function after all?

floesche, 2022-12-21:
The previous assumption is confirmed. If I move the file back to the previous location, I can run the `da.run_analysis` on the data sets.

taylorlm88, 2022-12-22:
Just confirming, yes, the single_fly_analysis.mat file which was created using create_settings_file() has paths in it indicating where to find all the data. You could update the paths in single_fly_analysis.mat directly using matlab, or you could update the variables in create_settings_file() to indicate your new location and run it to create a new settings .mat file. Alternatively, move the data back to the location the single_fly_analysis.mat expects it to be in.

closed 2023-10-12 by taylorlm88