JaneliaSciComp/G4_Display_Tools #68: error in setActiveAIChannels

floesche, 2023-02-06:
Received via slack, copying the error description here. I believe this error might be on floesche's branch [feature-panelscontroller](https://github.com/floesche/LED-Display_G4_Display-Tools/tree/feature-panelscontroller)

> I'm testing all the new run protocols and I'm having an issue with the streaming I thought you might be able  to  help with?
>
> In the streaming protocol, I use the command `ctlr.setActiveAIChannels(aibits);` to turn on the input channels. I was getting an error everytime I tried to submit the `startLog` command that looked like this:
>
>```matlab
>Error using  :
>Colon operands must be in the range of the data type.
>
>Error in PanelsController/expectResponse (line 1031) response = char(self.iBuf(pat.start(i):pat.end(i))); 
>Error in PanelsController/startLog (line 329) resp = self.expectResponse(0, 65, [], 10);
>Error in G4_default_run_protocol_streaming (line 193) log_started = ctlr.startLog();
>Error in G4_conductor_controller/run (line 711) eval(run_command);
>Error in G4_conductor_view/run_exp (line 621) self.con.run();
>Error in G4_conductor_view>@(varargin)self.run_exp(varargin{:}) (line 141) 'units', 'pixels', 'Position', [15, >self.fig_size(4)- 305, 115, 85],'Callback', @self.run_exp);
>```
>
> I figured out that this error was because pat.start was bigger than pat.end, and that it went away if I started the log running BEFORE doing the AI Channels command
> So I moved my command to start the AI channels to after teh start log command. But it turns out that no matter where I call the command to start the  AI channels, the very next controller command  returns a similar error. So now, when I try to submit the pattern ID for the pretrial,  I'm getting this error:
>```matlab
> Error using  +
> Integers can only be combined with integers of the same class, or scalar doubles.
> Error in PanelsController/expectResponse (line 1027) pat.end = pat.start + self.iBuf(pat.start);
> Error in PanelsController/setPatternID (line 404) resp = self.expectResponse([0 1], 3, [], 0.1);
> Error in set_controller_parameters (line 11) ctlr.setPatternID(p{2}); 
> Error in G4_default_run_protocol_streaming (line 223) set_controller_parameters(ctlr_parameters_pretrial);
> Error in G4_conductor_controller/run (line 711) eval(run_command);
> Error in G4_conductor_view/run_exp (line 621) self.con.run();
> Error in G4_conductor_view>@(varargin)self.run_exp(varargin{:}) (line 141) 'units', 'pixels', 'Position', [15, self.fig_size(4)- 305, 115, 85],'Callback', @self.run_exp);
>```

Slightly different but same general issue - in this case, pat.end is empty. I think it has to do with the fact that the buffer is not empty or something? Because  I am collecting data from it?

Actually it's not the first controller command after `ctlr.setActiveAIChannels(aibits);` that's returning an error, it's the second. The set control mode command is called directly before the pattern id command and it seems to work fine.

floesche, 2023-02-08:
Received via slack:

>  I still get the erro, but not always. The first time I ran it, I got this during the first intertrial:
>
> ```
> Error using  +
> Integers can only be combined with integers of the same class, or scalar doubles.
> Error in PanelsController/expectResponse (line 1029)
>                    pat.end = pat.start + uint64(self.iBuf(pat.start));
> Error in PanelsController/setControlMode (line 389)
>            resp = self.expectResponse([0 1], 16, [], 0.1);
> Error in set_controller_parameters (line 10)
>    ctlr.setControlMode(p{1});
> Error in G4_default_run_protocol_streaming (line 283)
>                    set_controller_parameters(ctlr_parameters);
> Error in G4_conductor_controller/run (line 711)
>            eval(run_command);
> Error in G4_conductor_view/run_exp (line 621)
>            self.con.run();
> Error in G4_conductor_view>@(varargin)self.run_exp(varargin{:}) (line 141)
>                'units', 'pixels', 'Position', [15, self.fig_size(4)- 305, 115, 85],'Callback', @self.run_exp);
>```
>
>It seems like `pat.start` is a `double 1x2` and can't be added to the `uint64`. However, I paused here at other times throughout the experiment and it seems like when `pat.start`  is a `double`, but only one number (like 1), it works fine.
>
> The second time I ran it, it went through the entire experiment with no errors.
>
> The third time I ran it, on the second trial I got the same error as before, and again `pat.start` is a `1x2 double`.

floesche, 2023-02-08:
I can't reproduce the issue at this point. 

Also, in the two most recent cases the error happens in a call from the function `set_controller_parameters()`. 

I don't know why this happens there, but I think this function is better integrated as a method into the controller object. That would allow easier debugging as well as more detailed access to the internal state of the controller. Currently this function looks like a workaround, duct taped to the object.

I'll need to pull your changes to my branch at some point to debug this.

In the meantime, building on your intuition of the `double` vs `uint64`, I added another type cast on my side with the this [commit](https://github.com/floesche/LED-Display_G4_Display-Tools/commit/b5f169063cba996c006b7ed11eecfc19c3d301cc). Now we shouldn't be getting an overflow error anymore…

taylorlm88, 2023-02-10:
By controller object do you mean the Panels controller or do you mean the Conductor controller? The run protocols have historically not been a method of the Conductor controller because they were meant to be user accessible for editing and customization, whereas the class is not. So I created a modules folder which has all the common code between all the run protocols in it as standalone functions which can be called by any run protocol (but also easily edited if desired). Since we are now providing a limited number of run protocols, rather than assuming the user will be accessing and customizing them, I could re organize it so that even the run protocols are controller methods if that'd be beneficial. 

The error only happens in this function because almost all the calls to the panel controller occur in this function, and it's the next panel controller command after some tiny delay that produces the error.

taylorlm88, 2023-02-15:
The error is fixed with the latest commit. Thank you! I will write a PanelsController method to replace my set parameters function.

closed 2023-10-12 by taylorlm88