JaneliaSciComp/G4_Display_Tools #76: Clicking on grayed out fields shows weird HTML tags

floesche, 2023-06-22:
If I click on a grayed out field, it allows me to go into an edit mode, but shows weird HTML tags. Only once I want to confirm the change, it tells me that this is not allowed. Is there a way to prevent the click before the user tries to enter data?

This is the case for clicking on Frame Rate, Gain, or Offset for Mode 1:

[image]

taylorlm88, 2023-06-25:
It was years ago that we set this up. At the time at least, matlab's table features were pretty limited and we decided it was not worth the amount of work it would take to fix this (if it was even possible, which we weren't sure it was without going under matlab and adding in some C code.) It's been a long time since then so I could go back and see if there are any new table functionalities that would solve this, because it would definitely be cleaner looking to not allow the user to click the cell at all. Will add this to the low priority issues to look into.

floesche, 2024-04-23:
Maybe this can be addressed at the same time as #26?

taylorlm88, 2024-07-23:
The code re-write to remove dependencies on java access has addressed this as well. It's finished, but not pushed to master yet pending thorough testing. Commit 368c1f87cedd72398d342aa523b12cc74b1f7581

closed 2024-07-23 by taylorlm88