JaneliaSciComp/G4_Display_Tools #77: Saving an experiment doesn't copy the Analog Output Function

floesche, 2023-06-22:
How to replicate:

- Start Experiment Designer
- Create a simple protocol with at least one Analog Output Function
- File -> Save As
  - choose new directory (in my case "c:\matlabroot\MyExperiment01"
  - Choose "MyExperiment01.g4p" as a file name
  - Click "Save"

[image]

The resulting folder `06-22-23_13-16-15` contains the subfolders `Analog Output Functions`, `Functions`, and `Patterns` along with `MyExperiment01.g4p` and `currentExp.mat`. `Functions` and `Patterns` have the `*.mat` and `*.pfn` / `*.pat` files, but `Analog Output Functions` is empty.

floesche, 2023-06-22:
In a second test with just one trial and 2 differently named AO functions, the export worked.

So maybe the problem seen above is related to either the frame index or using the same name for the Position and the Analog function? 

Needs more investigation.

taylorlm88, 2023-06-25:
In the first attempt when you had trouble saving, the AO function name is the same as the position function name - but is it actually the same file? Is the binary that goes with the function a .pfn or .afn file? Does it have the prefix fun or ao?

floesche, 2023-06-26:
No, it was not the same file. I created a Position function with that name first, which created a `*pfn` file in the `Functions` folder. Then switched over to AO functions and created an AO function with the same name, which created a `*afn` file in the `Analog Output Functions` folder.

In the designer I loaded the whole folder, which gave me different functions (and renderings) for functions/AO functions.

taylorlm88, 2023-10-02:
This happened to me yesterday when testing out the AO index display in the Conductor. If I imported the functions and created a new experiment with them and saved, it worked fine. But if I opened an existing protocol with AO functions, changed something, then saved it, the AO functions did not export. Will look into it this week.

taylorlm88, 2023-10-12:
Updating to add that I tried this on my computer (when it happened to me last week I was on Eyal's old computer) and I couldn't recreate the issue. No matter how I did it, the AO functions exported just fine. Will see if I can re-create on the other computer and if so, i'll investigate the difference.

floesche, 2024-04-23:
Probably fixed