JaneliaSciComp/G4_Display_Tools #80: Log Reps error in G4_TDMS_folder2struct

floesche, 2023-06-22:
When running a very simple protocol with one trial and either the `Streaming + Log Reps` or the `CC + Streaming + Log Reps` run protocol, the following error occurs:

```matlab
Error using G4_TDMS_folder2struct (line 61)
Unable to perform assignment because the indices on the left side are not compatible with the size of the right side.

Error in G4_conductor_controller/convert_multiple_logs (line 791)
                G4_TDMS_folder2struct(tdms_folder_path);

Error in G4_conductor_controller/run (line 718)
                self.convert_multiple_logs(fly_results_folder);

Error in G4_conductor_view/run_exp (line 602)
            self.con.run();

Error in G4_conductor_view>@(varargin)self.run_exp(varargin{:}) (line 140)
                'units', 'pixels', 'Position', [15, self.fig_size(4)- 305, 115, 85],'Callback', @self.run_exp);
 
Error while evaluating UIControl Callback.
```

Here the settings in the conductor:

[image]

taylorlm88, 2023-06-25:
Do you know what version of the host you were using to collect the data? Because this is the same error I get when data has been collected with v244 or later but I'm using the old folder2struct file. Unfortunately, when almost anything goes wrong in the folder2struct file, this is the error we get because it usually happens in a parfor loop, so hard to know if it's actually the same error or not. Just want to rule out version issue before delving further.

floesche, 2023-06-26:
I used v249. 

Yes, when trying to debug, I ended up in a `parfor` loop.

taylorlm88, 2023-09-09:
resolved with commit 1956fe9cddb5ae13e740f8daa4d46fbf68c4e6be

closed 2023-09-09 by taylorlm88