JaneliaSciComp/G4_Display_Tools #81: Conductor: flyingdetector error when with processing and plotting activated

floesche, 2023-06-22:
Trying to run a minimal protocol with a single trial and "processing" and "Plotting" activated, I see the following error:

```matlab
Error using load
Number of columns on line 13 of ASCII file C:\Users\labadmin\src\Modular-LED-Display\Generation 4\Display_Tools\G4_Data_Analysis\data processing\G4_Process_Data_flyingdetector.m must be
the same as previous lines.

Error in feedback_model/get_wbf_limits (line 224)
            S = load(processing_path);

Error in G4_conductor_controller/run (line 604)
                self.fb_model.get_wbf_limits(self.model.processing_file);

Error in G4_conductor_view/run_exp (line 602)
            self.con.run();

Error in G4_conductor_view>@(varargin)self.run_exp(varargin{:}) (line 140)
                'units', 'pixels', 'Position', [15, self.fig_size(4)- 305, 115, 85],'Callback', @self.run_exp);
 
Error while evaluating UIControl Callback.
```

Settings in the Conductor:

[image]

taylorlm88, 2023-06-25:
The default paths in the plotting and processing spaces are Matt's old code and don't work. For a protocol, you have to create a settings file for each specific to that protocol and load those files in the box before running the experiment. I will update the conductor's settings file so these paths are no longer there by default, because if the user doesn't remember to replace them with their own, they'll always getting processing and plotting errors.

taylorlm88, 2023-06-25:
commit 851bbcd

closed 2023-06-25 by taylorlm88