JaneliaSciComp/G4_Display_Tools #87: Display glitching

floesche, 2023-09-01:
All of a sudden, and without any known change to the system, we see glitches on at least 3 G4 arenas right now. Visually this means that we see garbled frames, often in one half of the arena while the other panels still show correct parts of the pattern. On the G4 Host, the green light "deque timeout" turns on the moment we see the issue. If we wait a bit longer, the "request FIFO TO" often comes on as well.

One example looks like this: 
[image]

We saw this issue with v242 and v247 at different setups. We since updated the setup that had the old v242 to v247 and continue to see the same problem. The configuration matches the size of the arena, 12×4 in our 3 cases.

If we look at the log files, we see many spikes in the timing, even before the display breaks visibly. Here is a typical example:
[image]

At this point I assume, that the visible error starts occurring when the frame length is shorter than 2ms (close to the 3rd to the last spike). 

A shorter example looks like this:
[image]

I ran one example with a very high buffer size (20000 for both, streaming and non-streaming GS16). In that example, it removes the intermediate spikes but still has the shorter than 2ms frames followed by a garbled display. This was just for debugging purposes, since 20000 is not acceptable for real closed loop experiments.
[image]

For reference, this is what the timing looked like before 8/25 (with v242):
[image]

The CPU usage is low, there is not a single core on that machine that gets utilized more than 20% during error.

The drive throughput is low, we don't see any spikes before or after the error. In fact, we tried to copy large files during the trial, but there is no correlation with seeing the problem.

There is no network traffic visible in the system monitor, except for the local traffic to G4 Host.

A suggestion in #48 was to look at the C-State in the BIOS. We did that an disabled them now, assuming that C-States can affect PCIe performance.

In #59 we received some tools to measure throughput. I ran the Bandwidth tester from https://github.com/floesche/LED-Display_G4_Display-Tools/releases/tag/Bandwidth-tester and saw a consistent 32MB/s throughput. I then ran the DMA consistency tester and saw many issues within a few minutes:

Here 16 errors within 4 minutes. Curiously they always seem to happen in pairs:
[image]

I am not sure if the DMA consistency checker uses the configuration of the G4 Host, but I saw the same type of issues within 2 minutes even when I increase the buffer size to 50000:
[image]

We currently focus our tests on one machine, but could collect similar debugging data from others if that seems helpful?

Are there other tests we can do?

floesche, 2023-09-01:
I ran another test without the wired network connected. It looks similar:

[image]

This is the resource monitor for network around the time when the glitching started:

[image]

achiusciotex, 2023-09-01:
The timeout errors are an indication of the data not being available when it is time to send it to the display.  Is there a way to monitor the IRQ (interrupt requests)? I think that has some impact on the application. 

The "DMA consistency checker" does not use the buffer size in the config file. I'm busy with something else right now so I might not be able to help much until next week, but I'll try to give suggestions on things to try as I think of them. Have you tried using the system on a fresh PC that has never been on the network or had updates pushed to it?

mbreiser, 2023-09-01:
Thanks @achiusciotex...yes at the moment we appreciate any suggestions on things to try. @floesche is now running tests on a fresh PC system with hopefully few updates....those results soon.  
---
but one question for you -- can you explain (maybe at a high level) what you think the difference is between the large frame intervals and the catastrophic glitches we've been seeing. Of course we don't want either, but an occasional longer frame interval we can live with, but the garbled frame data makes it impossible to do experiments.

achiusciotex, 2023-09-05:
@floesche the fact that you are not able to run the "Test DMA throughput consistency" exe without getting errors is problematic. The tester is basically setting up a buffer of 3 frames and the host side is putting in a frame's worth of data when space is available. The FPGA side is checking the DMA buffer every 2ms to see if there enough data for at least 1 frame, if not then it will flag an error. So for some reason, that DMA data is not getting there in time on your PC. I ran the tester on my PC for 20 minutes and had no errors.

I'll probably have to contact NI's tech support to see what can cause DMA transfer slow down/interruptions.

I'm curious, if you go to task manager and set the priority to high for the application does that change anything?

achiusciotex, 2023-09-06:
@floesche I contacted NI tech support about this issue and provided them with the tester code as well to see if they have an insight on the issue.

floesche, 2023-09-06:
Here some more observations to help @achiusciotex debug the issue:

I set up a completely new PC, installed a fresh Windows 10 Pro for Workstations, and set up the G4 Host using the installer from v230.

The Test DMA throughput consistency does not show any issue within the first 20 minutes.

I see the following issue:
- every _nth_ entry in the `*_Frame_Time.tdms` file has a very long delay compared to the previous entry (I have seen differences between 120000...340000)
- n differs depending on the protocol we run
  - for a protocol with alternating 200ms | 3000ms patterns, the issues are at entry 1000, 16000, 17000, 32000, 33000, 48000, 49000...
  - for a protocol with alternating 200ms | 2000ms patterns, the issues are at entry 1000, 11000, 12000, 22000, 23000, 33000, 34000...
- Issues happen in Control Mode 4 and Control Mode 1
- Issues happen at different times after initiating the `Start Display` command
  - for the 200|3000 protocol, the timing is 20754 after `Start Display` and 30087193 before the next `Set Control Mode`. For the next 3 issues the two times are: [20214, 2032253], [21857, 30058644], [19444, 2020278].
- The command sequence is: `Set Control Mode = 1` , `Set Pattern Id = 1`, `Set Pattern Function ID = 1`, `Set Position X = 1`, `Start-Display = 300`

Example plots of the time difference for the 200|3000 protocol (Mode 1, about 5 minutes) looks like this:
[image]

For the 200|2000 protocol (Mode 4, about 20minutes), it looks like this:
[image]

My observations:

I have seen the exact same issue with G4 Host v235, v242, and v247. So I don't think this relates to any recent bug fixes in the G4 Main Host.

The issue depends on frames, not time. This suggests that it is not a Windows problem - I would see some kind of drift over time. Whatever causes the issue knows exactly what the current frame number is - so it must be an internal G4 Main Host issue (or Labview / FPGA / other internal counters...)

On my current test machine, these issues do not trigger any of the previous mentioned "dequeue" or "FIFO" lamp in the G4 Host to light up. Over time the same issues seem to provoke the "dequeue" error on the "production" machine, which will then lead to the glitches in the display itself.

On the "production" machine, we did not see the issue before 8/25. It started happening between 8/25 and 8/28.

I will reset my test machine now.

Let me know what other information would be helpful to identify the problem within the G4 Main Host application.

achiusciotex, 2023-09-07:
@floesche can you send me your files that you used to test?

floesche, 2023-09-07:
These observations were based on running some of our protocols. I will try to recreate the same situation in a short script in the early afternoon today.

floesche, 2023-09-07:
Hi @achiusciotex , [here](https://github.com/floesche/LED-Display_G4_Display-Tools/blob/7ba651617aa53ceb1de72bb22f830aed8ed1c118/PControl_Matlab/test_scripts/PanelsControllerFunctionalTest.m#L131C1-L131C1) is a unit test that produced the error.

When I ran the test on my machine, I get this timing issue:

[image]

The errors occur exactly at the frames 15002 and 30002 in this log file:

[09-07-2023_10-10-00-15_Frame_Time.tdms.zip](https://github.com/JaneliaSciComp/G4_Display_Tools/files/12550067/09-07-2023_10-10-00-15_Frame_Time.tdms.zip)

achiusciotex, 2023-09-07:
@floesche  Can you give me all your pattern,function files, etc. in your matlabroot folder?

floesche, 2023-09-07:
Attached the `test` folder with the experiment (incl pattern + function files)
[test.zip](https://github.com/JaneliaSciComp/G4_Display_Tools/files/12552150/test.zip)

achiusciotex, 2023-09-07:
> 

@floesche Maybe I am not understanding what your issue is. You created the file before the run and started the display 3 separate times. The first 30 seconds show consistent timing. After the duration of the sequence is complete, it will continue to display the last frame, but will not log those frames in the log file. You see a 25 ms gap because you are now sending the commands again to restart the sequence. Again, after 30 seconds, you see a gap during the time you are sending commands. The display will always have a frame displayed, but stops logging after the sequence is finished.

floesche, 2023-09-07:
@achiusciotex OK, thanks. That explains the spikes in the observation. So the observations starting with [this comment](https://github.com/JaneliaSciComp/G4_Display_Tools/issues/87#issuecomment-1709109271) and the test case might not be able to help with the original issues.

The original issues was, that in some cases the frame is shorter than 2000. Example plot from the initial comment describing the error:

[image]

This will cause the display to [glitch](https://github.com/JaneliaSciComp/G4_Display_Tools/issues/87#issue-1877500401). We see this on 3 of our machines, apparently not on a clean new Windows.

Sorry for mixing up 2 separate issues. Any idea why a frame could be shorter?

floesche, 2023-09-07:
I wrote another simple script using the same pattern and function files that I shared earlier. Find the [`errorIssue87v2` here](https://github.com/floesche/LED-Display_G4_Display-Tools/blob/b00434d5f3710a39991cee572833f24a4788d671/PControl_Matlab/test_scripts/PanelsControllerFunctionalTest.m#L146)

Out of 2 runs, this failed twice with the following log files on our production machine.

[Log Files.zip](https://github.com/JaneliaSciComp/G4_Display_Tools/files/12553800/Log.Files.zip)

Each time I saw glitches on the display similar to the initial. This happens whenever the difference between frames is significantly shorter than 2000 and usually happens right after those longer frames at the beginning when a new start display command is sent to the G4 Host.

floesche, 2023-09-07:
The [initial unit test](https://github.com/floesche/LED-Display_G4_Display-Tools/blob/b00434d5f3710a39991cee572833f24a4788d671/PControl_Matlab/test_scripts/PanelsControllerFunctionalTest.m#L131) also produces the same error, as demonstrated in the attached Log files for `09-07-2023_18-06-38-45`.

[Log Files Mode1.zip](https://github.com/JaneliaSciComp/G4_Display_Tools/files/12553956/Log.Files.Mode1.zip)

achiusciotex, 2023-09-08:
@floesche I just ran the test 3 times without an issue. So to be clear, this is only a problem on the production machines and not the clean windows? I think this goes back to the issues we were seeing in the DMA consistency test we saw on your production machines. 

I was talking to a colleague about this and they did mention that a recent windows update caused some issues with drivers. So maybe that is what we are seeing with non-optimized hardware drivers causing interrupts. I think graphics card drivers in particular are affected more frequently. Maybe try updating those drivers on the production machine?

floesche, 2023-09-08:
I can try that. 

Here another test from another production machine that we just ran: 
[image]

What bothers me is that frames are sent with intervals <2000. How is that possible? Since the hardware is not capable of processing frames much faster than 2ms this must lead to issues. 

What could cause the G4 Host to send the frames in shorter intervals?

achiusciotex, 2023-09-08:
@floesche this is how the timing could lead to frames being sent faster than 2ms. Ultimately the issue is related to the data in the DMA not being available when it should.

[image]

achiusciotex, 2023-09-08:
I can make a build that will delay the output of the next frame to the next cycle if the previous frame was outputted late. You will not have consistent 2ms frames, but it will probably resolve the issue with the garbled frames. The DMA transfer issue will still be an issue.

floesche, 2023-09-08:
That would be great.

achiusciotex, 2023-09-08:
Test build for delaying frames if previous frame is late

https://www.dropbox.com/scl/fi/zt7619mon52290h9g3wq7/G4-Host-Ver1-0-0-251.zip?rlkey=p7i4ozszwel2qn6gb93jvci14&dl=0

floesche, 2023-09-08:
Thanks @achiusciotex . It tried version 251 but I still see frames shorter than 2000 and they still produce the same glitches.

Plot:

[image]

Example frames:
[image]

Log files:
[LogFiles_v251_01.zip](https://github.com/JaneliaSciComp/G4_Display_Tools/files/12563886/LogFiles_v251_01.zip)

I used the "errorIssue87" function and the patterns I shared previously.

achiusciotex, 2023-09-11:
@floesche I'll take another look at it. In the meantime, is it possible to set up fresh PCs so you can continue your experiments while we try to find out the root cause of these DMA transfer delays?

floesche, 2023-09-11:
Yes, I will do that today. We are running low on FPGA cards, so testing might include swapping cards and will be a bit difficult…

achiusciotex, 2023-09-11:
https://www.dropbox.com/scl/fi/jn79dy1fiu608gnd67xvn/G4-Host-Ver1-0-0-252.zip?rlkey=t77lvmgousybg7xkdls28kb6r&dl=0

You can try this version to see if it fixes the short frame issue. I also increased the DMA FIFO size of the FPGA side to see if this can help with the issue with transferring.

floesche, 2023-09-11:
Thanks, @achiusciotex. Your update seems to fix the issue. 

Just to double check: this version of the G4 Host is read for production (e.g. no background debugging is happening) and both, the 4bit and 2bit mode (500 and 1000 Hz) should work – correct?

achiusciotex, 2023-09-11:
@floesche I had ADC logging disabled in that version. This version should be ready for production, but test it out first. LabVIEW doesn't play well with merging files so creating a separate branch and merging it back together doesn't work all that well. So I re-enabled what I needed to and removed the debugging in this versin.

https://www.dropbox.com/scl/fi/edzexk8jz3v1ay4ekevzz/G4-Host-Ver1-0-0-253.zip?rlkey=mrb8l97dzxy3cqzhssncww2nk&dl=0

achiusciotex, 2023-09-11:
This may work, but I increased the size of the FPGA buffer so you will probably see more latency. The issue is still the transfer between host buffer and the fpga buffer being interrupted. I made the FPGA buffer 4 times as large, but we can experiment and see if we can go smaller without it failing. The flow of the image is backwards, but you get the point.
[image]

floesche, 2023-09-11:
Can you make the size of this buffer a parameter from the Config file? That would allow us to test and adjust according to the hardware at specific systems...

achiusciotex, 2023-09-12:
I need to set that size before I build the FPGA bitfile. For now, maybe I can build 4 different bitfiles for each bit mode (8 total) and allow you to select which one you want to run in the config file. But in the future, if we can narrow down the size that would be great because building 8 bitfiles for new FPGA releases can be bit time consuming.

floesche, 2023-09-12:
You couldn't set a parameter on which position the new element would enter the buffer, e.g. say 0 if I want to utilize the whole buffer, 6 if I only want to use a buffer of size 2?

Also, just to get the perspective right: the current buffer on the pattern reading side is 8 elements long, which basically translates to a 16ms latency for the 500Hz mode? Before you were using a 2 element buffer, which meant a 4ms latency – or am I getting the math wrong here?

Anyway, I would like to test the current G4 Host for a few days. Once we confirmed that the issue is really resolved across all our systems, maybe I could get back to you and request the 4 different files for the 4 bit mode?

achiusciotex, 2023-09-12:
It is a FIFO so you cannot control the position on the inserted data. 

There are two buffers, the FPGA and host buffer. The config file sets the host buffer size. If you are setting the buffer to 4000 then that is a little over 2 frames worth of data on the host side. The FPGA side used to hold around 1 frame of data. The reason I set it so small was because I thought the DMA Engine would be able to transfer that quickly, but now we are seeing inconsistencies so something seems to be interrupting it. The FPGA buffer in the build I released to you holds about 4 frames of data.

achiusciotex, 2023-10-24:
@floesche are you fine with the current fpga buffer size or do you want to test different sizes to minimize the latency?

floesche, 2023-10-24:
We currently don't have the bandwidth to test this, but will need to do this in the future.

achiusciotex, 2023-10-24:
Ok, I'll get the software ready for you to test it whenever you have time for it.

achiusciotex, 2023-10-30:
Added configurable FPGA frame size buffer. Valid values are 1-4
Parameter in ini file - FPGA FIFO Buffer Multiplier (1-4) = 4
https://www.dropbox.com/scl/fi/vw52sna9y1wrdapbhyk8l/G4-Host-Ver1-0-0-256.zip?rlkey=b7u6ofgupguax7k4x4donkgkc&dl=0