JaneliaSciComp/G4_Display_Tools #88: unrecognized field name "arena_pitch"

floesche, 2023-09-01:
When opening an experiment from another machine via Experiment Designer and then choosing to `Save As`, I see the following output:

```
G4_Experiment_Designer

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

ans =

     1219591

Unrecognized field name "arena_pitch".

Error in G4_document/export (line 1363)
                    newcurrentExp.pattern.arena_pitch(j) = self.Patterns.(field).pattern.param.arena_pitch;

Error in G4_document/saveas (line 1465)
            export_successful = self.export();% 0 - export was unable to complete 1- export completed successfully 2-user canceled and export not attempted

Error in G4_designer_controller/saveas (line 1370)
            self.doc.saveas(full_path, prog);

Error in G4_designer_controller>@(varargin)self.saveas(varargin{:}) (line 379)
            uimenu(menu, 'Text', 'Save as', 'Callback', @self.saveas);
 
Error while evaluating Menu Callback.

```

taylorlm88, 2023-09-09:
Were the patterns in the experiment you loaded created with the Pattern Generator? It seems like at least one of the patterns in the pattern folder was missing the arena_pitch field. I don't know how necessary that particular field is but we decided at one point a long time ago that patterns loaded to the software had to have all the fields created by the Pattern Generator, so if they're made another way, they need to have all the matching fields. Would need to reproduce the error and figure out which pattern it is and check to make sure it has all its fields. If it does, then this is definitely a bug, but if it is missing the field then I think this error is expected.  Unless we want to allow certain fields to be missing? I can test this out if you let me know what protocol it was.

floesche, 2024-04-23:
@leburnett sends list of required and optional parameters in the next month

floesche, 2024-04-23:
Copied from slack, posted by @taylorlm88 :

> These pictures show the fields currently expected in patterns and functions when they are imported. I gather most are not always necessary. Which fields, besides the actual data, do you want to be required, if any? Are there any that would cause problems if the user forgot to include them?
>
> [image]
> [image]

taylorlm88, 2024-05-07:
Fixed pending testing

closed 2024-05-09 by taylorlm88