JaneliaSciComp/G4_Display_Tools #92: time estimation for experiments seem weird

floesche, 2023-09-05:
In this specific example, the expected experiment length is 18m44s, the elapsed time at the end of the experiment is 18m55.65s, and the remaining time is -1m-11.65s. This is 1 min off, but also the double `-` suggests there is potentially something else wrong in the display?

floesche, 2023-09-06:
This offset by -1 min seems to be consistent, no matter how long the experiment is.

[image]

floesche, 2023-09-08:
In other cases there is no estimation:

[image]

Not sure if this is intended...

taylorlm88, 2023-09-09:
I just pushed a fix yesterday to fix a bug in the expected experiment length where some experiments had incorrect or wrongly formatted estimations. This issue may be improved by yesterday's change. The fix did not change how the remaining time is calculated, however. You do end up with a longer elapsed time and therefore negative remaining time when using features like pausing the experiment or re-running trials at the end that did not run correctly during the experiment. This is on my to do list, to make it more precise and take these features into account.

floesche, 2023-09-11:
Just to provide feedback: version 9fc7bd392e25599b791514dd0a1da92c8e59eb89 still shows the issue:

[image]

For most of the experiment the offset seems to be calculated correctly, but at the very end it somehow adds a "-1" to it...

taylorlm88, 2023-11-03:
This is fixed with commit 5ec8f50d78e10f2027a1f2d2f63fc157a1266655. Note that there will still be a small amount of "negative" remaining time at the end, and this amount will increase as the length of the entire experiment increases. I think it essentially adds up to all the time taken by code setting controller variables, updating graphics, and other things that happen in between trials. A possible improvement could be to find a way to account for these extra tiny chunks of time that add up. But the additional -1m should be gone now.

closed 2023-11-03 by taylorlm88