JaneliaSciComp/G4_Display_Tools #93: hidden error messages still open after Conductor is closed

floesche, 2023-09-05:
I missed closing an error window of the Conductor before closing the Conductor. In the MATLAB command window I then saw this error when closing the error message:

```matlab
Invalid or deleted object.

Error in G4_conductor_controller/create_metadata_file (line 1184)
                self.model.set_metadata_comments(self.view.comments_box.String);

Error in G4_conductor_controller/run (line 687)
            self.create_metadata_file();

Error in G4_conductor_view/run_exp (line 603)
            self.con.run();

Error in G4_conductor_view>@(varargin)self.run_exp(varargin{:}) (line 140)
                'units', 'pixels', 'Position', [15, self.fig_size(4)- 305, 115, 85],'Callback', @self.run_exp);
 
Error while evaluating DestroyedObject Callback.
```

Is there a way to stop closing a window while there are still error messages open? Or keep the error messages in the foreground? or close child windows when closing the main window?

taylorlm88, 2023-10-02:
I introduced a closing function in commit 5d44013e6ccc88df1cd1a1753d23aa0f6de90570 which I think may resolve this but I haven't confirmed yet because I haven't replicated the issue. Will test in the future to see if it still happens.

taylorlm88, 2024-05-28:
So there's no way to make the error dialog boxes a child of the main GUI (and therefore have them close when the GUI does) but I can set them to be 'modal' which means you will not be able to interact with the GUI (including closing it) until you have dealt with the error box, so that will prevent any from accidentally being missed.

taylorlm88, 2024-05-28:
Resolved with commit 31805bb

closed 2024-05-28 by taylorlm88