JaneliaSciComp/G4_Display_Tools #96: Show name of protocol in conductor

floesche, 2023-09-06:
Once the G4P is loaded, there is no indication in the Conductor about the protocol name that I can see. 

Would it be nice to add the g4p file name somewhere on the GUI?

taylorlm88, 2023-09-09:
Does it not show up in the Experiment Name field on the right hand side? It's greyed out, so it can't be changed. That also means if it's a long name you can't scroll right to see all of it, but you do see at least the first portion of the file name.

floesche, 2023-09-11:
Oh, yes. I see it now.

Having a more prominent place could help. In Windows it's often in the title, in this case behind "Fly Experiment Conductor".

Otherwise, being able to scroll and see the whole name would also be good.

taylorlm88, 2023-09-29:
Added in commit 67c8077

closed 2023-09-29 by taylorlm88