JaneliaSciComp/G4_Display_Tools #99: Error when not loading pattern step size from file

floesche, 2023-09-07:
After clicking "Load from pattern" and then cancelling the loading in the file window, the Function Generator produces the following error.

```matlab
Error using G4_Function_Generator_gui>pushbutton7_Callback
Expected pattern metadata not found in selected file

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in G4_Function_Generator_gui (line 42)
    gui_mainfcn(gui_State, varargin{:});

Error in matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)G4_Function_Generator_gui('pushbutton7_Callback',hObject,eventdata,guidata(hObject))
 
Error while evaluating UIControl Callback.

Error using G4_Function_Generator_gui>pushbutton7_Callback
Expected pattern metadata not found in selected file

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in G4_Function_Generator_gui (line 42)
    gui_mainfcn(gui_State, varargin{:});

Error in matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)G4_Function_Generator_gui('pushbutton7_Callback',hObject,eventdata,guidata(hObject))
 
Error while evaluating UIControl Callback.

```

This is probably not used very often. But if this happens, can I keep using the function generator or not?

taylorlm88, 2023-09-09:
Doing some quick testing, it looks like if you get this error, the function generator continues to work as expected. I can create and save functions after receiving that error without any issues as far as I can tell. I will look into that callback function, should be a simple fix to this bug.

taylorlm88, 2023-09-09:
Resolved with commit 9fc7bd3

closed 2023-09-09 by taylorlm88