JaneliaSciComp/G4_Display_Tools #107: Remove references to panel_com from G4 documentation

floesche, 2024-04-25:
Panel_com is retired, but the documentation still references it. For example:

- https://reiserlab.github.io/Modular-LED-Display/Generation%204/Display_Tools/docs/experiment-conductor_run-protocol_tutorial.html#the-panel_com-wrapper
- https://reiserlab.github.io/Modular-LED-Display/Generation%204/Display_Tools/docs/pcontrol.html#the-panel_com-command (the commands are useful, maybe just rewrite this page to reflect current script names)
- example code, e.g. in https://reiserlab.github.io/Modular-LED-Display/Generation%204/Display_Tools/docs/experiment-conductor_run-protocol_tutorial.html#run-the-conditions

taylorlm88, 2024-04-25:
Thanks for finding these. I had not bothered updating the run protocol tutorial because I was planning to remove it. I missed the file about the panel com command. I will get these updated and make sure there are no other references out there.

taylorlm88, 2024-05-02:
Completed with commit 8f27e9a

closed 2024-05-02 by taylorlm88