JaneliaSciComp/G4_Display_Tools #108: Add an example experiment to the repository

floesche, 2024-05-01:
Currently, the tests we have to see if a new setup works are limited to very basic functionality (on / off). After that, we have very nice tutorials that helps people to get started with setting up their own experiments.

I think, it would be nice to have some basic experiments in the repository that are ready to run. They might even be the files that are generated in the tutorials.

taylorlm88, 2024-05-01:
Yes, we can create a suite of experiments to include as tests/examples. Will work on this.

taylorlm88, 2024-05-07:
Is there a reason test_protocol1 (found in G4_Display_Tools/G4_Protocol_Designer/test_protocols/test_protocol1) doesn't fit this bill? It is a ready to run experiment that users can play before an actual experiment by hitting the "run test protocol" button on the Conductor. But they can also open it by browsing to it and run it like a regular experiment as well. It has four mode 1 conditions and mode 4 pre/inter/posttrial conditions.  Do we want different example experiments showcasing different modes or combinations of pre/inter/post trials?

floesche, 2024-05-08:
I think it would be nice to have that explained as part of the initial verification process, for example as an additional step at https://reiserlab.github.io/Modular-LED-Display/Generation%204/Display_Tools/docs/software_setup.html#verify? Currently it is somewhat hidden in https://reiserlab.github.io/Modular-LED-Display/Generation%204/Display_Tools/docs/protocol-designer_configure-settings_tutorial.html#the-settings and I had completely missed this when I was looking for a simple way to verify that a setup was working.

taylorlm88, 2024-05-08:
Yes, it was initially intended to be used when running an experiment so it never made it to the set up docs, but it can easily be used for both. I will create a second test protocol for a 3 screen arena (pretty sure test_protocol1 is for 4 rows), and then put them both in the setup docs.

taylorlm88, 2024-05-09:
I created a second test protocol for 3 screen arenas and renamed them so it's clear which is which. I updated the software setup docs to include steps to make sure the Designer and Conductor are working as well as to run a test protocol, both with and without the Conductor GUI. Let me know if I missed anything regarding this issue, closing it for now. commit 639dc2d

closed 2024-05-09 by taylorlm88