JaneliaSciComp/G4_Display_Tools #109: Document what versions of MATLAB are supported

floesche, 2024-05-01:
I can't find information about what versions of MATLAB are working.

There seem to be big changes in 2012 and again in 2016. Do we support earlier versions? What versions are tested?

taylorlm88, 2024-05-01:
I would need to go back and test. All versions since I started building it in 2019 should work, I don’t remember any issues that weren’t corrected on our end. But versions previous to that haven’t been tested. I can look into doing this.

floesche, 2024-05-02:
It might be good enough to say "R2019b and newer" as long as we know it works with the most recent MATLAB as well…?

taylorlm88, 2024-05-07:
I upgraded to Matlab 2024a and everything seems to work as expected.

taylorlm88, 2024-05-07:
I added to the readme a line stating that the tools are compatible with matlab versions R2019b and newer, but are untested on earlier versions. If you think this is enough, I'll go ahead and close this issue, but I could download earlier versions and test them if you think it's worth it, especially if there are particular versions that underwent large changes.

floesche, 2024-05-07:
Adding the versions [here](https://reiserlab.github.io/Modular-LED-Display/Generation%204/Display_Tools/docs/software_setup.html#download-matlab) might also be helpful? Maybe 2019b…2024a instead of 2019b and newer?

I think it is more than reasonable to support 5 years and 10 versions of a language and software development kit, so no need to test earlier versions from my point of view.

taylorlm88, 2024-05-07:
Sure, we'll just need to remember to update the docs every time a new version is released, once we test it out. I'll add to that page as well.

taylorlm88, 2024-05-07:
resolved with commit 246b814

closed 2024-05-07 by taylorlm88