JaneliaSciComp/G4_Display_Tools #110: Should run_experiment_G4_example.m work?

leburnett, 2024-05-01:
Should the file `\G4_Display_Tools\G4_Example_Experiment_Scripts\run_experiment_G4_example.m` work? 

I tried running it earlier but got the following error:
[image]

It appears that the folder `C:\matlabroot\G4\Experiments\` doesn't exist anymore. @floesche and I just pulled the newest code onto the computer so the files should be up to date.

taylorlm88, 2024-05-01:
Probably not, it was created years ago. I can update it so it will work again, and get some improved examples.

taylorlm88, 2024-05-07:
@leburnett FYI there is an example experiment at "G4_Display_Tools\G4_Protocol_Designer\test_protocols\test_protocol1" if you are looking for a sample experiment to run. I think the script 'run_experiment_G4_example.m' was created even before we used run protocols or anything.

leburnett, 2024-05-07:
> @leburnett FYI there is an example experiment at "G4_Display_Tools\G4_Protocol_Designer\test_protocols\test_protocol1" if you are looking for a sample experiment to run. I think the script 'run_experiment_G4_example.m' was created even before we used run protocols or anything.

Thanks, @taylorlm88 I can give it a try and check that everything runs smoothly.

taylorlm88, 2024-05-09:
@floesche @leburnett 

The files in G4_Example_Experiment_Scripts show how to create an experiment completely with scripts, bypassing the Designer and Conductor all together. They were written before we had the Designer and Conductor to create and run experiments. The file run_experiment_G4_example.m doesn't work until you use the file create_experiment_G4_example.m to create one. 

Do we want to keep these files? If we want these files to be available so users can see programmatically what goes into an experiment, or even so users can use them if they prefer programming to using the GUIs, I can update them so they will work again. But if we don't want users to have the option of creating and running experiments with these scripts, I can just remove them all together. 

What do you guys think?

floesche, 2024-05-09:
How much effort would be required for updating them? 

I think there is a good use case for keeping them functional, but if it requires two weeks of work, that use case goes away.

taylorlm88, 2024-05-14:
I've updated these scripts and they are working again. You have to run "create_experiment_G4_example.m" first, then "run_experiment_G4_example.m"

taylorlm88, 2024-05-14:
The scripts work and i've updated the software setup documentation. commit d4647e2 and commit 4cdf910

closed 2024-05-14 by taylorlm88